rerun-io/rerun #10761: Scale3D mono index out of bounds

CameronMatthew, 2025-07-28:
**Describe the bug**

`Scale3D` throws a warning instead of doing scaling.

```python
import rerun as rr

print(rr.__version__)
# 0.23.4

rr.init("scale_example", spawn=True)

rr.log("example_a", rr.Transform3D(scale=rr.Scale3D(2.0)), static=True)
# Couldn't read /example_a:rerun.components.Scale3D @ ((TimeInt::STATIC, RowId(185663C1D827AF633839bc73d55cfe71))): mono index out of bounds: 0 (len=3)

rr.log("example_b", rr.Transform3D(scale=rr.Scale3D([2.0, 2.0, 2.0])), static=True)
# Couldn't read /example_b:rerun.components.Scale3D @ ((TimeInt::STATIC, RowId(185663C1D82971E93839bc73d55cfe73))): mono index out of bounds: 0 (len=3)
```

**To Reproduce**
Run the minimal reproducible example above^.

**Expected behavior**

**Backtrace**

```[STACKTRACE]
Couldn't read /example_a:rerun.components.Scale3D @ ((TimeInt::STATIC, RowId(18566434E368E7AE1FDB8E2BF5CFDBA6))): mono index out of bounds: 0 (len=3)
```

**Desktop (please complete the following information):**
 - OS: macOS Sequoia 15.5

**Rerun version**

- Originally discovered on 0.23.4. I was able to reproduce from 0.22.0 upwards.

grtlr, 2025-07-29:
Aha, so we would implement something like `Vec3DLike` for the `Scale3D` component?

ntjohnson1, 2025-07-29:
`Scale3D` actually defers to `Vec3D` which also doesn't have a length property. I think for anything that is plural with a single field we should automatically generate the length property then I think our existing logic works. (I'm basically just copying how we handle the  automatic `__array__` generation.

```python
from rerun.datatypes import Vec3D
>>> a = Vec3D([1,2,3])
>>> len(a)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: object of type 'Vec3D' has no len()
```

closed 2025-07-30 by ntjohnson1