rerun-io/rerun #10787: Support attaching custom data to each pixel in an image

emilk, 2025-08-04:
Suppose you have an RGB image, and for each pixel you want to attach some custom data, such as:

* label: string
* confidence: f32
* …

How would you do it? I see three different ways of doing it:

## A) Log an extra entity per image
Log an extra `Boxes2D` entity, instanced so that you end up with one box per pixel. You can then attach custom components to that entity.

This is the ugliest solution, but also the only one that works reliably at the moment.

Pros:
* Works today
* Supports any and all component types (you can even attach an image to each pixel of an image, e.g. for light-field visualization)

Cons:
* https://github.com/rerun-io/rerun/issues/10276
* A lot of tiny boxes cover your image
* Hacky
* Requires a separate child entity

## B) Log an extra image per custom data
Log multiple images to sibling entities, one for each type of custom data.

For instance, you could log `confidence` as a `Luminance` image with datatype `f32`.

Pros:
* Works today
* Pro: you can visualize the confidence image with color mapping

Cons:
* `Image` is limited in the types it supports (each pixel can only be L, RGB, or RGBA, of u8, u16, f32 etc)
  * No support for strings (but can often be done with `SegmentationImage`)
  * No support for e.g. a jacobian per pixel
  * No support for custom components
  * Has wrong semantic meaning (encoding "confidence" as "luminance" doesn't really make sense)
* Our support for stacking dozens of images on top of each other is poor (but can be fixed)
  * We don't show all of them when hovering
  * What image ends up on top is not well-defined
  * It's difficult to configure which image ends up on top
  * It's difficult to configure the transparency of each image
* `Image`s are not really made to be vessels of custom data
  * we could use a `Tensor` instead, but it's not a lot better

## C) Add support for "projecting" components onto pixels
The user would log a single 640x480 RGB image, together with 307200 values for a custom "confidence" component, and a 307200 strings with labels, etc. They would log these components to the same entity.

If the user hovers the pixel at (X=52, Y=23) we translate that to pixel number 52+23*640=14772, and then look up the values of the 14772:th instance of all components at this entity path, and show that in the tooltip when hovering an image.

Pros:
* Supports all custom components
* Easy to implement

Con
* Slightly hacky
* This would be slightly nicer if we supported nullable components, which we want to remove:
  * https://github.com/rerun-io/rerun/issues/6819

Related:
* #1756

teh-cmc, 2025-08-04:
> What happens if the number of components is different from WxH?

Clamping semantics have worked wonders for us 🤷‍♂️

chengguizi, 2025-08-30:
Some thoughts on Option B.

A workaround I used to use is to log multiple DepthImage data, which come with ability to set draw order. This is helpful if the value I need per pixel visualisation is in float and make sense to be visualise in a gradient coloring format.

It is a pity that the **current Tensor Archetype cannot be treated as image and viewed by Spatial2DView.** That is semantically more correct, rather than hijacking the DepthImage type and repurpose it for visualising float.

Right now, in latest rerun 0.24.x versions (or more accurately since https://github.com/rerun-io/rerun/pull/9775), I can **no longer log multiple DepthImage** on top of a single image and let the 2D view automatically overlay them (it worked in 0.20.0):
- Can this limitation be lifted somehow, so that we can truly log and visualise multiple custom per pixel data on a single image?
- If not lifted on DepthImage, perhaps add tensor-over-image support would be great.

Arguably the changed behaviour https://github.com/rerun-io/rerun/pull/9775 will prevent proper implementation of option B, as option B requires multiple depth/tensor data to be logged on one color image

> Now we only overlap images if they are the same resolution, AND there is at most ONE color image (including videos) and ONE depth image.

I think there is a good fix: If the heuristics can make use of user loggable `rerun::archetypes::Transform3D::INVALID` to tell `foo` and `foo/bar` should not be visualised on the same view, such 1 color + 1 depth/tensor limitation could be lifted.

Does this line of thought make sense?

chengguizi, 2025-08-30:
@Wumpf Yes, totally get point on the part of limitation regarding 1 DepthImage and 1 Color. The point I want to make here, that same limitation make Option B not possible.

I believe option B assumes we can log multiple Image and overlay them. Or use Tensor which requires more code change which make is render-able on Spatial2DView.

Generally curious on how this might move forward towards 0.26? "Attaching custom data to each pixel" is generally quite a useful thing after all!