rerun-io/rerun #10796: C++ SDK compilation fails with CMake > 4.0

SebastianoTaddei, 2025-08-05:
**Describe the bug**

`rerun_cpp_sdk` fails to compile on macOS with CMake > 4.0.

Since CMake 4.0, compatibility with CMake < 3.5 has been removed. That means that any project with `cmake_minimum_required(<3.5)` will fail to compile. That is the case for some dependencies of arrow (as tracked [here](https://github.com/apache/arrow/issues/45985)) and consequently makes Rerun fail to compile.

**Workaround**

Until arrow updates its dependencies, Rerun can be compiled by setting the environment variable `export CMAKE_POLICY_VERSION_MINIMUM=3.5`. It would be nice to be able to pass it directly as a CMake flag via `-DCMAKE_POLICY_VERSION_MINIMUM=3.5`, but that is somehow not forwarded by Rerun (as mentioned briefly on [Discord](https://discord.com/channels/1062300748202921994/1075873257124810852/1401910471656734752)).

**To Reproduce**

Steps to reproduce the behavior:

1. Download and extract the artifact: [sdk](https://github.com/rerun-io/rerun/releases/download/0.24.0/rerun_cpp_sdk.zip).
2. Compile manually with: `cmake -Bbuild` and `cmake --build build`.
3. Compilation fails when compiling arrow (see screenshot and backtrace for the error).

The same goes if you try to use Rerun via `FetchContent` (for example when compiling the dna example).

**Expected behavior**

`rerun_cpp_sdk` should compile.

**Screenshots**

[image]

**Backtrace**

[arrow_cpp-build-out.log](https://github.com/user-attachments/files/21591403/arrow_cpp-build-out.log)

Relevant log:

```log
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.
```

**Desktop:**

 - OS: macOS Sequoia 15.5
 - CMake: 4.0.3

**Rerun version**

0.24.0, taken directly from the release page

**Additional context**

I have some colleagues reporting this on Rerun 0.22.0 as well. It probably affects all Rerun versions.

nikolaseu, 2025-09-15:
Hey I don't think it's worth to reopen this issue, at least not from my side, it's an arrow issue. And it's easily solved by setting the corresponding environment variable instead of the CMake variable.

ntjohnson1, 2025-09-15:
> Hey I don't think it's worth to reopen this issue, at least not from my side, it's an arrow issue. And it's easily solved by setting the corresponding environment variable instead of the CMake variable.

I mostly re-opened so I can take a look. Either adding the env variable in our function scope with a note that it is happening, or at least providing a clearer error message.

PR #10799 (2025-08-05): Resolve CPP Arrow Build
### Related
* closes #10796 

### What
This checks for newer CMake and automatically applies an automatic workaround.
If a user explicitly provides a `CMAKE_POLICY_VERSION_MINIMUM` we defer to that.

Verified by running the cmake commands from `cpp-compile-all`
```console
cmake -G 'Ninja' -B build/debug -S . -DCMAKE_BUILD_TYPE=Debug
cmake --build build/debug --config Debug --target ALL
```
This works

```console
cmake -G 'Ninja' -B build/debug -S . -DCMAKE_BUILD_TYPE=Debug -DCMAKE_POLICY_VERSION_MINIMUM=4.0
cmake --build build/debug --config Debug --target ALL
```
This  [... cut]