rerun-io/rerun #10812: C++ SDK linking error on macOS

SebastianoTaddei, 2025-08-06:
**Describe the bug**

`rerun_cpp_sdk` linking fails on macOS. This issue was mentioned and resolved [here](https://github.com/rerun-io/rerun/pull/8991), but I believe it is still present in one of the CMake compilation approaches.

In particular, if used with `find_package` it fails to link.

The problem is due to the need to link `-framework Security` when linking against the sdk.

**Workaround**

For now a simple workaround is to link against rerun like this:

```cmake
target_link_libraries(example_dna PRIVATE rerun_sdk "-framework Security")
```

**To Reproduce**

Steps to reproduce the behavior:

1. Create a random folder and cd into it.
2. Follow the steps to get and compile the sdk explained [here](https://github.com/rerun-io/rerun/blob/main/rerun_cpp/cmake_setup_in_detail.md#cmake-install).
3. Use the sample `main.cpp` and `CMakeLists.txt` snippets.
5. Compile with `cmake -Bbuild` and `cmake --build build`.

`main.cpp`:
```cpp
#include <rerun.hpp>

int main()
{
  const auto rec = rerun::RecordingStream("rerun_example_dna_abacus");
  rec.spawn().exit_on_failure();

  return 0;
}
```

`CMakeLists.txt`:
```cmake
cmake_minimum_required(VERSION 3.16...3.27)

project(example LANGUAGES CXX)

set(rerun_sdk_DIR "${CMAKE_CURRENT_SOURCE_DIR}/rerun_cpp_sdk/lib/cmake/rerun_sdk/")
find_package(rerun_sdk REQUIRED)

add_executable(example main.cpp)

# Link against rerun_sdk.
target_link_libraries(example PRIVATE rerun_sdk)
```

**Expected behavior**

Code should compile.

**Desktop (please complete the following information):**

 - OS: macOS Sequoia 15.5

**Rerun version**

0.24.0, fresh from the artifact

**Additional context**

Unless I have made mistakes in following your guidelines, I believe it should just work as expected. Please let me know if I'm wrong.

SebastianoTaddei, 2025-08-07:
@grtlr thanks for pointing that out, using the development release of Rerun compiles as expected!

closed 2025-08-07 by SebastianoTaddei

grtlr, 2025-08-07:
Thank you for testing!