rerun-io/rerun #10835: Support timeline control and range selection events in the native viewer

wbthomason, 2025-08-07:
**Is your feature request related to a problem? Please describe.**
#8673 added support for selecting timelines and timepoints in the web viewer through the JS API. However, it does not currently appear possible to do the same for the native viewer through Rust.

**Describe the solution you'd like**
It would be useful to have API-level control of the timeline at least through Rust, if not the other SDK languages. Ideally, this would include support for registering callbacks for timeline events **including** range selection (e.g., shift+drag selection), extending the existing support for time update events and timeline selection events.

**Describe alternatives you've considered**
Duplicating the timeline somehow in an extended GUI - but then it's unclear if I could sync the state with the "real" timeline in a native viewer...

**Additional context**
This is intended for use in building a data annotation app extending the Rerun GUI, wherein I need to be able to select sub-ranges of a recording and send these time bounds to my extension app for further processing.

Wumpf, 2025-08-08:
> This is intended for use in building a data annotation app extending the Rerun GUI, wherein I need to be able to select sub-ranges of a recording and send these time bounds to my extension app for further processing.

if you're already part of the Rust process it shouldn't be too hard to inject timeline controls from outside into application. But just like Jan on Discord, of the bat I don't see this being exposed 🤔
A custom view would definitely already have access to the timeline btw.

wbthomason, 2025-08-08:
Thanks, I hadn't found #7728! Can you say more about a custom view having access to the timeline already? Do you mean because functions in `ViewClass` have access to a `ViewerContext` object? I'm not against adding a custom view if needed - but the rest of my UI is fairly complex and isn't well-suited to a view, and I'm not sure how a custom view would work when I mostly just want to use the default Rerun views and extract information from them for my custom UI.