rerun-io/rerun #10843: Video play broken with time ranged links

Wumpf, 2025-08-08:
`pixi run rerun-release 'rerun://sandbox.redap.rerun.io:443/dataset/1859951D49D9F54F053110352a835641?partition_id=ILIAD_49fe161f_2023_06_27_15h_52m_49s&time_range=real_time@2023-07-13T04:41:06.567641088Z..2023-07-13T04:41:15.114329088Z'`

This correctly fetches ~70MB / 16GB from this RRD. But the Video player gives a warning about "out of order samples" and then refuses to play.
https://rerunio.slack.com/archives/C07Q6HLG24S/p1754652218751459

(this rrd is merged together from a lot of different RRDs)

Could be that we have to improve how we handle out of order arriving samples. To be investigated what actually fails here

emilk, 2025-09-09:
There are two different issues to worry about:
* Individual chunks being "unordered" (not all timelines increasing monotonically with RowId)
* Chunks arriving out-of-order

I _believe_ the "out of order samples" refers to the first of these problems, and could probably be fixed by sorting the chunk by the active timeline.

IsseW, 2025-09-16:
Additional information from a deep-dive about this today:
- This issue is not specific to time ranged links and happens when just opening the recording manually as well.
- Chunks themselves are in order, looking at component time ranges. And samples within individual chunks are ordered. But in some cases different chunks' time ranges overlap, causing the first sample from a later chunk be earlier than the last sample in the earlier chunk. Which causes the warning, and that entire chunk to be ignored.

The working theory now is that since these recordings are merges of many different recordings video streams from multiple sources might've ended up at the same time. Causing the video not to load.

It might also be possible to detect that this specific case is happening and display a better warning for it.

IsseW, 2025-09-19:
Gotten some more insight into what's happening with this bug now. And there are two parts to it.
One part is the overlapping video like we thought, and that matches up with some of my debugging. Where two or more of the merged recordings were in the same time frame.

The other part of this is that for streamed videos the order in which the files were merged in matters. And for the streamed videos to work you have to merge the recording from an earlier time stamp before the one with the earlier timestamp.

So this will work:
`rerun rrd route --recording-id test Oct__7.rrd Oct_20.rrd > merged.rrd`
This won't work:
`rerun rrd route --recording-id test Oct_20.rrd Oct__7.rrd > merged.rrd`

Given that the same entity path in both recordings both have a video stream.

This might be something we want to support? And I think this is possible to fix . One option being invalidating the cache whenever we receive a chunk before the current video sample range. Or maybe it'd be possible to append frames at the start of the video.

Also had the idea if these videos could somehow be different cache entries, but doesn't feel correct to do it like that.
I think for now I'll just improve the error here and note this down in an issue.