rerun-io/rerun #10879: Dataframe Column Names Don't Consider Archetypes Leading to Collisions

ntjohnson1, 2025-08-12:
**Describe the bug**
Related to what caused me to see https://github.com/rerun-io/rerun/issues/10877 but when I tried to repro something closer to my real use case it felt like two separate bugs.

If I have this in an rrd the dataframe api cannot differentiate the two `data` entries. Even when one is static and the other isn't
[image]

**To Reproduce**
```python
import rerun as rr

rr.init("duplicate_tagged_components", spawn=False)
rr.save("sample_any.rrd")
# These are two groupings of content with a variety of fields but at least one intersects
schema = rr.AnyValues().with_field(rr.ComponentDescriptor("data", "mcap.Schema"), 1.0)
image = rr.AnyValues().with_field(rr.ComponentDescriptor("data", "SomeCustomImage"), 2.0)
rr.log("/camera/data", schema, static=True)
rr.log("/camera/data", image)
```

```python
>>> import rerun as rr
>>> rec = rr.dataframe.load_recording("sample_any.rrd")
>>> rec.schema()
Column name: /camera/data:data
        Entity path: /camera/data
        Archetype: SomeCustomImage
        Component type: 
        Component: data
Column name: property:RecordingInfo:start_time
        Entity path: /__properties
        Archetype: rerun.archetypes.RecordingInfo
        Component type: rerun.components.Timestamp
        Component: RecordingInfo:start_time
>>>batches = rec.view(index=None, contents="/camera/**").select()
print(next(iter(batches)))
pyarrow.RecordBatch
log_tick: int64
log_time: timestamp[ns]
/camera/data:data: list<item: double>
  child 0, item: double
----
log_tick: [null]
log_time: [null]
/camera/data:data: [[1]]

>>>batches = rec.view(index="log_time", contents="/camera/**").select()
<stdin>:1: RerunWarning: RecordingView::select: tried to select static data, but no non-static contents generated an index value on this timeline. No results will be returned. Either include non-static data or consider using `select_static()` instead.
```

**Expected behavior**
Internally we disambiguate these. So we should be able to expose that to the dataframe API.
Note that this has a python example but we see the same bug if we use a similar behavior in rust.

**Rerun version**
rerun-cli 0.25.0-alpha.1+dev (map_view nasm native_viewer) [rustc 1.88.0 (6b00bc388 2025-06-23), LLVM 20.1.5] aarch64-apple-darwin (debug)

ntjohnson1, 2025-08-13:
Plan is for me to deprecate `with_field` since the component identifier is only based on the field name and not the archetype which is unintuitive/ leads to duplicated shadowed entries in the rrd we can't query out.

@Wumpf will look at the fact that the viewer in the screenshot above doesn't actually represent the rrd faithfully since the archetype shouldn't differentiate things if the component identifiers are the same.

Wumpf, 2025-08-14:
Split out the viewer part to https://github.com/rerun-io/rerun/issues/10903