rerun-io/rerun #10997: Improve error handling in the buid scripts

YohDeadfall, 2025-08-25:
**Describe the annoyance**

Originally made as an `TODO` by @emilk:

https://github.com/rerun-io/rerun/blob/46cb497b050246221ae26f67347e741c5afe9f3a/crates/build/re_types_builder/src/codegen/cpp/mod.rs#L403

The build scripts are more mature now with `cargo_error` from #10941, issues are reported one by one which may degrade developer experience if something goes wrong and it requires a several iterations of fixes and compilations.

**To Reproduce**

Not applicable at the moment.

**Expected behavior**

All errors should be delivered using `cargo:error` or `cargo::warning`.

**Your goals**

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Any

**Additional context**

There's also a reporter type which is used in some cases, but not others. Perhaps, it can be eliminated too.

Wumpf, 2025-08-25:
the reporter type is only used in codegen which isn't used inside of build scripts - not everything inside the `crates/build` is about build.rs scripts!

YohDeadfall, 2025-08-25:
Then it means that everything should go through the reporter. Will fix the description.

Wumpf, 2025-08-27:
the reporter is specific to the codegen tool. But yes there almost everything (except fatal errors) should go to the error reporter