rerun-io/rerun #11028: LargeBinary Migration Doesn't Update Datafusion Schema

ntjohnson1, 2025-08-28:
**Describe the bug**
If we've already registered a dataset on the cloud with `list<list<uint8>>` in the schema then when we try to access that column the schema expects the original type but we are returning `list<large_binary>`

**To Reproduce**
Steps to reproduce the behavior:
1. Launch local notebook from cloud repo and point to existing dataset
2. Query a column with `blob` type
```python
>> dataset.dataframe_query_view(index="real_time", contents="/thumbnail/camera/wrist").df().limit(1)
Exception: DataFusion error: Execution("Arrow error: Invalid argument error: column types must match schema types, expected List(Field { name: \"item\", data_type: List(Field { name: \"item\", data_type: UInt8, nullable: false, dict_id: 0, dict_is_ordered: false, metadata: {} }), nullable: true, dict_id: 0, dict_is_ordered: false, metadata: {} }) but found List(Field { name: \"item\", data_type: LargeBinary, nullable: true, dict_id: 0, dict_is_ordered: false, metadata: {} }) at column index 4")
```

**Expected behavior**
We can query data even if migrating the format.

**Rerun version**
```console
rerun-cli 0.25.0-alpha.1+dev (base map_view nasm native_viewer oss_server release_no_web_viewer) [rustc 1.88.0 (6b00bc388 2025-06-23), LLVM 20.1.5] aarch64-apple-darwin (debug)
Video features: av1 default ffmpeg nasm serde
```
commit: 241df80b8d97f144aaefde067656257b712a9859

emilk, 2025-08-28:
In theory the fix is to pass the schema through `re_sorbet::migrate_record_batch`… in theory

jleibs, 2025-08-28:
> In theory the fix is to pass the schema through re_sorbet::migrate_record_batch… in theory

This might work for some trivial situations, but I believe it is significantly more complicated than that. There is an interplay between the data schema, as returned by the schema registration and the derived partition-manifest-schema which captures the temporal index structures needed for queries to behave properly.

closed 2025-09-02 by andrea-reale