rerun-io/rerun #11047: OSS Server Shows Loading Spinner Forever

ntjohnson1, 2025-08-29:
**Describe the bug**
When launching the OSS Server it initially select the host name and shows the spinner forever. If the down arrow is selected to select the actual dataset it is already loaded.

**To Reproduce**
Steps to reproduce the behavior:
1. rerun server -d <path to rrd>
2. rerun
3. command palate to add redap server (http, localhost, 51234)

**Expected behavior**
I think my expectation is that the dataset is selected at load and the the host isn't selectable or maybe shows something more than forever spinner.

**Screenshots**

[image]
[image]

**Rerun version**
```console
rerun --version
rerun-cli 0.25.0-alpha.2 (base map_view nasm native_viewer oss_server release_full release_no_web_viewer web_viewer) [rustc 1.88.0 (6b00bc388 2025-06-23), LLVM 20.1.5] aarch64-apple-darwin release-0.25.0-alpha.2 38a9647, built 2025-08-29T09:16:40.98495Z
Video features: av1 default ffmpeg nasm serde
```

Wumpf, 2025-08-30:
Afaik it's because the oss server doesn't support tables yet and it can't advertise this so far

abey79, 2025-10-30:
This has been fixed by @timsaucer's recent work on tables

closed 2025-10-30 by abey79