rerun-io/rerun #11290: URDF Loader does not respect scale parameter of meshes

germanrodriguez-mimic, 2025-09-19:
**Describe the bug**
If you load a urdf whose mesh objects have a scale different than `1,1,1`, the scaling is not done correctly and the meshes are displayed in their original size.

**To Reproduce**
- log a urdf normally
- log a urdf after chaging the scale parameter
```python
robot = URDF.from_xml_file(self.__urdf_path)
for link in robot.links:
    if link.visual is not None:
        geom = link.visual.geometry
            if geom.scale is not None:
                geom.scale = [2.0, 2.0, 2.0]
rr.log_file_from_contents(
           urdf_path,
           robot.to_xml_string().encode("utf-8"),
           static=True,
)
````

**Expected behavior**
The meshes have different sizes depending on the scale

**Desktop:**
 - OS: Ubuntu 22.04

**Rerun version**
0.25.1

**Additional context**
Many STLs are in mm so they need `scale=0.001 0.001 0.001`  to be properly displayed as URDFs

oxkitsune, 2025-10-06:
I'm unable to reproduce this.

We handle scale for mesh geometry here, which is correct.
https://github.com/rerun-io/rerun/blob/a151c79b5d3a2aa5ffaaa1e075802ef8e7da3f49/crates/store/re_data_loader/src/loader_urdf.rs#L568-L578

oxkitsune, 2025-10-08:
https://discord.com/channels/1062300748202921994/1418723693541986375
per discord, currently unable to reproduce this, leaning towards the collision geometry showing up in the place of visual geometry.
Removing this from release blocker list for 0.26, and closing this until we can reproduce.

closed 2025-10-08 by oxkitsune