rerun-io/rerun #11351: Can't run the viewer on macOS 26? No wgpu adapters found

emilk, 2025-09-26:
`pip install rerun-sdk && rerun` ->

```
[2025-09-26T15:44:29Z INFO  egui_wgpu] No wgpu adapters found
[2025-09-26T15:44:29Z ERROR eframe::native::run] Exiting because of error: WGPU error: No suitable graphics adapter found; noop not requested, vulkan support not compiled in, metal found no adapters, dx12 not requested, gl support not compiled in, webgpu not requested
Error: WGPU error: No suitable graphics adapter found; noop not requested, vulkan support not compiled in, metal found no adapters, dx12 not requested, gl support not compiled in, webgpu not requested
(base) 
```

---

Works: `pixi run rerun`

`cargo run -p rerun-cli` -> No wgpu adapters found

Also affects egui:
`cargo run` -> No wgpu adapters found
`cargo run -p egui_demo_app` -> SEGFAULT (!)

Also affet wgpu repo:

`cargo run --bin wgpu-examples hello_triangle` ->

```
Failed to find an appropriate adapter: NotFound { active_backends: Backends(METAL), requested_backends: Backends(NOOP | VULKAN | GL | METAL | DX12 | BROWSER_WEBGPU), supported_backends: Backends(NOOP | METAL), no_fallback_backends: Backends(0x0), no_adapter_backends: Backends(METAL), incompatible_surface_backends: Backends(0x0) }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Wumpf, 2025-09-26:
We also reproed this this with a [minimal swift-written](https://github.com/ctreffs/metal-app/) Metal application iff that application was started directly from commandline. Starting the app bundle via `open` however would work just fine (those run presumably in a different environment?)

emilk, 2025-09-30:
FOUND IT!

I had set `DYLD_LIBRARY_PATH` in my `.zshrc`, which… somehow made it all fail.

This will make it A LOT easier to debug next time:
* https://github.com/emilk/egui/pull/7572

closed 2025-09-30 by emilk